ms-iot/imx-iotcore #10: USB devices get not recognized without external USB Hub

UweSteinkohl, 2019-01-10:
I am using a KaRo i.mx6DL board on a KaRo standard evaluation kit. I have the problem that USB devices like mouse or keyboard only work, if I connect them via an USB hub.
This happens on the USB host port as well as on the USB OTG. THe USB hub does not need to be powered.

If I login via remote power shell, devcon status reports the following in the case it fails with no hub:
USB\VID_0000&PID_0001\5&312A3217&0&1
    Name: Unknown USB Device (Port Reset Failed)
    The device has the following problem: 43

and the following if connected with hub:
USB\VID_046D&PID_C00C\6&447B0C4&0&2
    Name: Logitech USB WheelMouse
    Driver is running.

does somebody knows a solution for this problem?

christopherco, 2019-01-10:
Hi,

Let's focus first on the USB Host issue. USB OTG failure could be a separate issue altogether.

> USB\VID_0000&PID_0001\5&312A3217&0&1
> Name: Unknown USB Device (Port Reset Failed)
> The device has the following problem: 43

Port Reset Failed is an early step in the USB device initialization sequence. You also do not get a valid VID/PID, which indicates early device descriptor transfer failure (or we haven't gotten to that stage yet). I suspect that the USB device is not getting adequate VBUS from the USB port when directly connected. This would cause the observed symptoms. And a USB-powered hub will circumvent the missing VBUS.

Please check that the USB port connected to the USB Host is outputting 5V on the VBUS line.

UweSteinkohl, 2019-01-11:
Hi,

thank you for your answer.

Yes, this actually was my first thought and I checked the power on VBUS. That was ok (4,9V).
On the other hand I think the VBUS cannot be the problem, because it works perfect with an unpowered hub.

perhaps some effects with the overcurrent? but as far as I can see the overcurrent detection is switched off in the UEFI USBPhy init.

christopherco, 2019-01-12:
Thanks for checking VBUS and agree it is not the issue here.

To further isolate the issue, can you check if the USB behavior is reproducible in UEFI shell and in U-Boot shell (both with and without hub)? This data will help determine if the issue is USB power management related or not.

U-boot and UEFI's EHCI driver always operates in D0 powered on state and is generally optimized for simple transactions, while Windows could attempt to power manage the controller and possibly put the HW in a bad state. 

If the issue is still present in U-boot or UEFI shell, I would suspect a firmware config issue issue, like a USB PHY misconfiguration.

christopherco, 2019-01-12:
Also, as an added update: I'm seeing some USB enumeration anomolies on my i.MX6DL Hummingboard Edge here. Specifically only one of the ports is not enumerating low speed devices unless it is plugged behind a hub. Currently investigating...

UweSteinkohl, 2019-01-14:
you are right, I have not seen that. Only low speed devices fail. All my USB sticks, SD card readers etc. work without hub.

I checked it in u-boot. there I have exactly the same results. low speed does only work behind a hub, whereas full speed works directly. 

Looking at the register bits of the USBPHYx_CTRLn, I would say a cleared ENUTMILEVEL2 bit would cause the problem. But I am sure this bit is set, so it should work.

christopherco, 2019-05-21:
I found the issue here. We were not advertising an internal Transaction Translator block to Windows. So Windows would only enumerate High-Speed devices. After advertising the internal TT block (via _HRV=1 in USB Device node), I was able to enumerate and use low-speed devices in Windows. See this PR for where I added _HRV https://github.com/ms-iot/imx-edk2-platforms/pull/31

Background on _HRV: https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/other-acpi-namespace-objects

Note, this change will only enlighten Windows that a TT is available. It will not address the U-boot enumeration issue

UweSteinkohl, 2019-05-21:
Hi Christopher,

 

thank you for pointing me to that solution.

 

regards

 

Uwe

closed 2019-06-06 by christopherco