ms-iot/imx-iotcore #15: Firmware build path issue under WSL

dvescovi1, 2019-01-17:
Clean install of WSL on my Windows 10 (build 1809).
Followed firmware build instruction verbatim for HummingBoardEdge_iMX6Q_2GB

following error:
.
.
.
OK
make[2]: Leaving directory '/home/david/edk2/BaseTools/Tests'
make[1]: Leaving directory '/home/david/edk2/BaseTools'
~ ~/imx-iotcore/build/firmware/HummingBoardEdge_iMX6Q_2GB
Loading previous configuration from /home/david/edk2/Conf/BuildEnv.sh
WORKSPACE: /home/david
EDK_TOOLS_PATH: /home/david/edk2/BaseTools
CONF_PATH: /home/david/edk2/Conf
Copying $EDK_TOOLS_PATH/Conf/build_rule.template
     to /home/david/edk2/Conf/build_rule.txt
Copying $EDK_TOOLS_PATH/Conf/tools_def.template
     to /home/david/edk2/Conf/tools_def.txt
Copying $EDK_TOOLS_PATH/Conf/target.template
     to /home/david/edk2/Conf/target.txt
Build environment: Linux-4.4.0-17763-Microsoft-x86_64-with-Ubuntu-18.04-bionic
Build start time: 16:43:42, Jan.17 2019

WORKSPACE        = /home/david
PACKAGES_PATH    = /home/david/edk2:/home/david/imx-edk2-platforms
EDK_TOOLS_PATH   = /home/david/edk2/BaseTools
CONF_PATH        = /home/david/edk2/Conf

Architecture(s)  = ARM
Build target     = RELEASE
Toolchain        = GCC5

Active Platform          = /home/david/imx-edk2-platforms/Platform/SolidRun/HummingBoardEdge_iMX6Q_2GB/HummingBoardEdge_iMX6Q_2GB.dsc
Flash Image Definition   = /home/david/imx-edk2-platforms/Platform/SolidRun/HummingBoardEdge_iMX6Q_2GB/HummingBoardEdge_iMX6Q_2GB.fdf

Processing meta-data .

build.py...
/home/david/imx-edk2-platforms/Platform/SolidRun/HummingBoardEdge_iMX6Q_2GB/HummingBoardEdge_iMX6Q_2GB.dsc(100000019): error 000E: File/directory not found in workspace
        ArmPkg/Library/DefaultExceptionHandlerLib/DefaultExceptionHandlerLibBase.inf is not found in packages path:
        /home/david/edk2
        /home/david/imx-edk2-platforms

- Failed -cd
Build end time: 16:43:43, Jan.17 2019
Build total time: 00:00:01

../Common.mk:195: recipe for target 'uefi' failed
make: *** [uefi] Error 14cd 
david@HOME5-PC:~/imx-iotcore/build/firmware/HummingBoardEdge_iMX6Q_2GB$
 

I can clearly see ArmPkg directory and it contents. Seems to something with paths?

christopherco, 2019-01-17:
A couple days ago Tianocore edk2 core removed DefaultExceptionHandlerLibBase.inf and caused a build break across all ARM platforms. They fixed it this morning (https://github.com/tianocore/edk2/commit/00755e1c8019cb073de85dea97737aedc10f352c) but there are some regressions in our edk2-platforms code surrounding this change. I'm currently working on addressing it.

If you want an immediate workaround to build, I recommend checking out tianocore/edk2 commit ID: 1f7b748315440e18ef85f0b076c8fd5e7b84259d and rebuild.

christopherco, 2019-01-18:
I have root caused the regression to another breaking commit in Tianocore edk2 (https://github.com/tianocore/edk2/commit/2f4a5a9f4c17ed88aaa3114d1e161e42cb80a9bf). From the issue's [Bugzilla](https://bugzilla.tianocore.org/show_bug.cgi?id=1426), this is a known regression and there is a patch which should hopefully be submitted soon.

I've added a PR to imx-edk2-platforms to address the initial DefaultExceptionHandlerLibBase build break.