ms-iot/imx-iotcore #16: firmware build against u-boot vs Solidrun u-boot

dvescovi1, 2019-01-20:
I am seeing several issues with my u-boot firmware (SabreLite) built against the ms-iot\u-boot vs. what was working under the private ms-iot\kayla\SolidRun-u-boot.
Instead of bringing the private \kayla\SolidRun-u-boot to public preview under something like ms-iot\imx-u-boot (which would seem to fit the new folder naming convention) it looks like you started with an older existing u-boot before several of the bootloader changes were incorporated. I am suspecting several of the platforms may also have the same problems?? 
Going forward are we targeting this repo for the imx-iotcore bootloader changes?
Should I re-PR the changes to this repo now?

jordanrh1, 2019-01-22:
Hi David,
The ms-iot/u-boot repository is the u-boot repository we'll use going forward. We collected the functionality into a minimal set of patches and applied them to a recent stable release of u-boot. If there were changes in the Solidrun-u-boot repository they will need to be brought over to u-boot.
What is the nature of the issue you're seeing?

dvescovi1, 2019-01-23:
most issues had to do with loading the "uefi.fit" file which was a name change from what they use to use.
Looks like the whole boot script uefi.fit loading was reworked as one of the last commits under SolidRun u-boot. Some other minor issues with config support. I submitted a new pull request to ms-iot/uboot and I have not seen any issues so far. I am concerned if you rebuild the u-boot firmware for some of the other "standard" platform you may also see the same issues. 
Do you normally rebuild all the platform's firmware as part of the C.I.?
Are they regression tested?

jordanrh1, 2019-01-23:
When we make firmware changes we run 'make' in the build/firmware directory which rebuilds all the platforms listed in the Makefile. It looks like SabreLite is not in the Makefile so that could explain why there's a break. It may have been deliberately removed. @hbeberman do you know the status of our SabreLite support?

hbeberman, 2019-01-23:
I've merged pull https://github.com/ms-iot/u-boot/pull/3. With this U-Boot change I was able to successfully build the SabreLite firmware using imx-iotcore.

@dvescovi1 Please rebuild and test the firmware by running `make` in /build/firmware/SabreLite_iMX6Q_1GB. Once you've tested the firmware you can stage a commit with the updated binaries by running `make update-ffu` in /build/firmware/SabreLite_iMX6Q_1GB.

Once you've opened a pull with updated SabreLite firmware binaries I'll add the board back into the documentation, firmware makefile, solution dependencies, FFU build script, and FFU build CI.

We do not currently run CI on U-Boot or OP-TEE pulls, but we do for EDK2-Platforms. When making updates to the firmware we regression test on the platforms we have on hand.

dvescovi1, 2019-01-23:
Sounds good. I am preparing the PR to ms-iot/iotcore for the full SabreLite support and everything looks good for the firmware so far.
My concern was that maybe some of the "other" platforms may have the same issue if someone tries to rebuild the firmware as it appears ms-iot/u-boot repo looks like it was slightly behind the ms-iot/SolidRun u-boot else it would have picked up my last SolidRun u-boot commit.

closed 2019-01-24 by dvescovi1