ms-iot/imx-iotcore #29: i.MX8 support

dvescovi1, 2019-02-13:


dvescovi1, 2019-02-13:
What is the status of i.MX8 support? I see some remnants.
Looking for updated ETA.
Willing to work on this, just wondering what repos the firmware was built from? (edk/uefi/u-boot etc.)

hbeberman, 2019-02-13:
We currently have a board package for NXP's i.MX8M EVK checked into the tree https://github.com/ms-iot/imx-iotcore/tree/public_preview/build/board/NXPEVK_iMX8M_4GB.

We're working to publish the new repositories used to build this firmware along with updated build instructions within the next few weeks.

dvescovi1, 2019-02-14:
I was able to build and flash an ARM64 SD card image using the prebuilt firmware.
U-boot boots ok but it stall in uefi:

ProtectUefiImageCommon - 0xFC485CC0
                                           
  - 0x00000000FCE6C000 - 0x0000000000016000
                                   
SetUefiImageMemoryAttributes - 0x00000000FCE6C000 - 0x0000000000001000 (0x000000
0000004008)
                                                                   
SetUefiImageMemoryAttributes - 0x00000000FCE6D000 - 0x0000000000012000 (0x000000
0000020008)
                                                                   
SetUefiImageMemoryAttributes - 0x00000000FCE7F000 - 0x0000000000003000 (0x000000
0000004008)
                                                                   
Tpm2DeviceLibOpTEEEntry: Entry
                                                
Tpm2AcpiControlAreaInit: About to write control area.
                         
TEEC[E]:gBS->LocateProtocol(gEfiRpmbIoProtocolGuid) failed. (Status=Not Found) (
OpteeRpcCmdRpmb: /mnt/d/github/project_mu/MU_PLATFORM_NXP/Microsoft/OpteeClientP
kg/Library/OpteeClientApiLib/Arm/OpteeClientRPC.c, 322)
                       
TEEC[E]:TEEC_SMC_OpenSession() failed. (TEEC_OpenSession: /mnt/d/github/project_
mu/MU_PLATFORM_NXP/Microsoft/OpteeClientPkg/Library/OpteeClientApiLib/OpteeClien
tApiLib.c, 340)
                                                               
Tpm2RequestUseTpm: OpenSession returned 0xFFFF3024, 0x3
                       
ASSERT [Tcg2Dxe] /mnt/d/github/project_mu/MU_PLATFORM_NXP/Microsoft/OpteeClientP
kg/Library/Tpm2DeviceLibOptee/Tpm2DeviceLibOptee.c(178): ((BOOLEAN)(0==1))
    
                                     

any ideas? Looks like TPM issue?

hbeberman, 2019-02-14:
It looks like UEFI is failing to find the RPMB protocol because the MMC driver didn't register one. The fTPM has a hard dependency on RPMB as its secure storage device, and RPMB is an eMMC only feature.

What device are you running the FFU on?
Does it have an eMMC?

dvescovi1, 2019-02-15:
this is the standard NXP i.MX8M EVK.
Right out of the box. Just changed jumpers to boot from uSD slot.
I am booting from uSD card. Does it have to be programmed in to eMMC flash?

neilsh-msft, 2019-02-15:
No it doesn't need to be programmed to eMMC flash.  The iMX8M EVK board we're using has SW802 set to '1 0', and SW801 set to '1 1 0 0' for SD card boot.  The part number of the board we're using is 170-29615 REV B2 as etched in copper on the underside of the board, but has a sticker SCH-29615 REV B3.

dvescovi1, 2019-02-15:
I rebased and generated image from public_preview as of 2-13.
I have rev SCH-29615 B4.
switches set the same.

dvescovi1, 2019-02-16:
also have 700-29615 REV A2 marking.

dvescovi1, 2019-02-17:
I was going to attempt to build a "non-secure" version of uefi.fit (for the time being) to bypass this issue but it appears this uefi was generated differently. Looks like it used project mu. Not having much luck. I do not see and "platform" specific code in mu. Any advice would be appreciated.

dvescovi1, 2019-02-22:
@neilsh-msft Was able to rebuild and boot a "non-secure" version uefi.fit on i.MX8MQ on EVK.
working so far.

closed 2019-02-22 by dvescovi1