ms-iot/imx-iotcore #31: SYSTEM_THREAD_EXCEPTION_NOT_HANDLED on ClSomImx7

jordanrh1, 2019-02-18:
Some time after the first boot, Compulab IMX7 enters a reboot loop where it bugchecks just after the Windows logo appears. The bugcheck is `SYSTEM_THREAD_EXCEPTION_NOT_HANDLED`. It occurs after bootmgr and before the debugger attaches for NT kernel, which probably means it's happening in winload. I have been unable to attach the boot debugger to winload, so unable to gather more information. Will likely need JTAG to debug.

jordanrh1, 2019-02-18:
It was booting into UpdateOS, and UpdateOS was crashing. Enabled kernel debugging of UpdateOS with:

```
bcdedit /store k:\efi\Microsoft\boot\bcd /debug {311b88b5-9b30-491d-bad9-167ca3e2d417} on
```

You can enumerate all boot entries with:
```
bcdedit /store k:\efi\Microsoft\boot\bcd /enum all
```

The bugcheck is:
```
	BugCheck 7E, {c0000709, 8c05a068, 8cc03880, 8cc03680}
	
	Probably caused by : mx6pep.sys ( mx6pep!MX6_PEP::InitializeResources+12c )

```

mx6pep should not be loaded on IMX7!

jordanrh1, 2019-02-19:
It also looks like HalExtiMX7Timers is not present, which later causes a bugcheck when it tries to go into low power mode.

jordanrh1, 2019-02-19:
@hbeberman @neilsh-msft  It looks like the wrong UpdateOS is getting picked up for IMX7. An IMX6 UpdateOS is being included in IMX7 images. Can one of you take a look?

hbeberman, 2019-02-20:
Issue should be resolved by pull https://github.com/ms-iot/imx-iotcore/pull/33. The board specific cabs changed directories and the override to grab the correct UpdateOS cab was looking in the wrong location.
i.MX7 FFUs now contain the correct UpdateOS drivers.

closed 2019-02-20 by hbeberman