ms-iot/imx-iotcore #37: u-boot rebuild ARM64 i.mx8 evk

dvescovi1, 2019-02-26:


dvescovi1, 2019-02-26:
I was attempting to rebuild u-boot for the i.MX8MQ per the new re-build instructions.
I am using WSL.
I can clone and configure no problem but rebuild stalls at the SPL.
I think we have seen something similar in the past when we rebuilt this way.
I don't recall just what the issue was ...something with WSL name path support possibly?
Anyone recall?

here is dump:
...
  LD      spl/drivers/usb/dwc3/built-in.o
  CC      spl/fs/fs_internal.o
  LD      spl/fs/built-in.o
  LDS     spl/u-boot-spl.lds
  LD      spl/u-boot-spl
  OBJCOPY spl/u-boot-spl-nodtb.bin
  COPY    spl/u-boot-spl.bin
  CFGS    arch/arm/mach-imx/spl_sd.cfg.cfgtmp
  MKIMAGE SPL
./tools/mkimage: Can't open SPL: Invalid argument
arch/arm/mach-imx/Makefile:162: recipe for target 'SPL' failed
make[1]: *** [SPL] Error 1
Makefile:1115: recipe for target 'SPL' failed
make: *** [SPL] Error 2
david@HOME5-PC:/mnt/c/ms-iotMX8/u-boot$

dvescovi1, 2019-02-27:
Pretty easy to reproduce. just checkout u-boot beta branch under wsl.
install dependences and compiler. 
Do the two exports to set ARCH=arm64 and the cross compiler then do:
make imx8mq_evk_nt_defconfig
make

dvescovi1, 2019-03-01:
@neilsh-msft 
Seems to be quite a few issues rebuilding MX8 ARM64 firmware under wsl.
Under VM it works a little better … I can get through u-boot build but still getting error building optee.

dvescovi1, 2019-03-01:
@neilsh-msft 
optee os is building.
optee trusted app is failing.

dvescovi1, 2019-03-04:
error seem to be cenetred around defines in the wolfssl build and the file Implementation.h ... umong other things.

dvescovi1, 2019-03-04:
also it seems strange we are building the ARM32 sample app?  Per instructions:
"pushd ~/ms-tpm-20-ref/Samples/ARM32-FirmwareTPM/optee_ta"

??

neilsh-msft, 2019-03-04:
For building, SPL it may be an issue with file name case sensitivity. I've been building it in WSL from the path /home/<user name>/... which has case sensitive paths and not from /mnt/<drive name>/... which are case insensitive.

As with the ARM32 sample app, it builds fine for ARM64.  MS Research hasn't renamed the folder yet after we asked them to support ARM64 as well.  However OpTEE does build with a fair number of warnings if the 7.2.1 GCC compiler is used.  Using 6.4.1 has much fewer warnings, but I haven't tested much with that build of GCC.  

The build error you're seeing with Implementation.h is an issue with the latest set of commits pushed to the TPM repo.  The Implementation.h was updated for the core code, but not for the sample TPM code.  Please checkout commit 'd979cae6db01ea2e57dbce92d700ce1d0edf78fa' to get a successful build.

Thank you for your time and patience.

dvescovi1, 2019-03-04:
Neil:
Rebuilding all now after making the minor adjustments.

Thanks for the help!

closed 2019-03-04 by dvescovi1