ms-iot/imx-iotcore #38: Problem to build imx-iotcore and problem to load uefi.fit

awietholter, 2019-02-27:
Hello, I have two problems:
First case: compile imx-iotcore
Second case: Run uefi.fit.

**First case:**
I'm using a proprietary board, based on the Sabre_iMX6Q_1GB platform.
I made the necessary changes to u-boot as indicated in the porting-imx.md document.
https://github.com/ms-iot/project-kayla/blob/develop/Documentation/porting-imx.md
However, I can not compile the UEFI if I change data other than the Sabre_iMX6Q_1GB card standard.
imx-iotcore/build/firmware/myboard_iMX6Q_1GB/Makefile
For example:

```
UBOOT_CONFIG=myboard_nt_defconfig

EDK2_DSC=myboard_iMX6Q_1GB
EDK2_PLATFORM=mycompany/myboard_iMX6Q_1GB
EDK2_DEBUG_RELEASE=RELEASE

UBOOT_OPTEE_ITS=uboot_optee_unsigned.its
UEFI_ITS=uefi_unsigned.its

all: firmware_fit.merged firmwareversions.log

include ../Common.mk

.PHONY: $(OPTEE)
$(OPTEE):
	$(MAKE) -C $(OPTEE_ROOT) O=$(OPTEE_OUT) PLATFORM=imx-mx6qsabresd \
	 `$(OPTEE_FLAGS_IMX6)
```

The error appears after compiling u-boot and optee.
Follow:

```
WORKSPACE        = /home/test
PACKAGES_PATH    = /home/test/edk2:/home/test/imx-edk2-platforms
EDK_TOOLS_PATH   = /home/test/edk2/BaseTools
CONF_PATH        = /home/test/edk2/Conf
PYTHON_COMMAND   = /usr/bin/python3.6

Architecture(s)  = ARM
Build target     = RELEASE
Toolchain        = GCC5

Active Platform          = /home/test/imx-edk2-platforms/Platform/test/myboard_iMX6Q_1GB/myboard_iMX6Q_1GB.dsc
Flash Image Definition   = /home/test/imx-edk2-platforms/Platform/test/myboard_iMX6Q_1GB/myboard_iMX6Q_1GB.fdf

Processing meta-data ....... done!
Building ... /home/test/edk2/ArmPlatformPkg/MemoryInitPei/MemoryInitPeiLib.inf [ARM]
Building ... /home/test/edk2/ArmPlatformPkg/PlatformPei/PlatformPeiLib.inf [ARM]
make[1]: Entering directory '/home/test/Build/myboard_iMX6Q_1GB/RELEASE_GCC5/ARM/ArmPlatformPkg/MemoryInitPei/MemoryInitPeiLib'
"arm-linux-gnueabihf-gcc" -mthumb -march=armv7-a -g -Os -fshort-wchar -fno-builtin -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -include AutoGen.h -fno-common -mlittle-endian -mabi=aapcs -fno-short-enums -funsigned-char -ffunction-sections -fdata-sections -fomit-frame-pointer -Wno-address -mthumb -mfloat-abi=soft -fno-pic -fno-pie -fstack-protector -mword-relocations -flto -Wno-unused-but-set-variable -Wno-unused-const-variable -Dmyboard_iMX6Q_1GB -DCPU_IMX6DQ -DSOCTYPE_QUAD -Wno-unused-local-typedefs -DCONFIG_DUMP_SYMBOL_INFO=1 -DCONFIG_MPCORE=1 -DCONFIG_USB=1 -DCONFIG_OPTEE=1 -DCONFIG_AUTH_VAR=1 -DCONFIG_MEASURED_BOOT=1 -DCONFIG_SECURE_BOOT=1 -nostdinc -nostdlib -DUEFI_C_SOURCE -mno-unaligned-access -c -o /home/test/Build/myboard_iMX6Q_1GB/RELEASE_GCC5/ARM/ArmPlatformPkg/MemoryInitPei/MemoryInitPeiLib/OUTPUT/./MemoryInitPeiLib.obj -I/home/test/edk2/ArmPlatformPkg/MemoryInitPei -I/home/test/Build/myboard_iMX6Q_1GB/RELEASE_GCC5/ARM/ArmPlatformPkg/MemoryInitPei/MemoryInitPeiLib/DEBUG -I/home/test/edk2/MdePkg -I/home/test/edk2/MdePkg/Include -I/home/test/edk2/MdePkg/Include/Arm -I/home/test/edk2/MdeModulePkg -I/home/test/edk2/MdeModulePkg/Include -I/home/test/edk2/EmbeddedPkg -I/home/test/edk2/EmbeddedPkg/Include -I/home/test/edk2/ArmPkg -I/home/test/edk2/ArmPkg/Include -I/home/test/edk2/ArmPlatformPkg -I/home/test/edk2/ArmPlatformPkg/Include /home/test/edk2/ArmPlatformPkg/MemoryInitPei/MemoryInitPeiLib.c
<command-line>:0:1: error: macro names must be identifiers
Building ... /home/test/edk2/ArmPlatformPkg/Library/ArmPlatformStackLib/ArmPlatformStackLib.inf [ARM]
make[1]: Entering directory '/home/test/Build/myboard_iMX6Q_1GB/RELEASE_GCC5/ARM/ArmPlatformPkg/PlatformPei/PlatformPeiLib'
"arm-linux-gnueabihf-gcc" -mthumb -march=armv7-a -g -Os -fshort-wchar -fno-builtin -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -include AutoGen.h -fno-common -mlittle-endian -mabi=aapcs -fno-short-enums -funsigned-char -ffunction-sections -fdata-sections -fomit-frame-pointer -Wno-address -mthumb -mfloat-abi=soft -fno-pic -fno-pie -fstack-protector -mword-relocations -flto -Wno-unused-but-set-variable -Wno-unused-const-variable -Dmyboard_iMX6Q_1GB -DCPU_IMX6DQ -DSOCTYPE_QUAD -Wno-unused-local-typedefs -DCONFIG_DUMP_SYMBOL_INFO=1 -DCONFIG_MPCORE=1 -DCONFIG_USB=1 -DCONFIG_OPTEE=1 -DCONFIG_AUTH_VAR=1 -DCONFIG_MEASURED_BOOT=1 -DCONFIG_SECURE_BOOT=1 -nostdinc -nostdlib -DUEFI_C_SOURCE -mno-unaligned-access -c -o /home/test/Build/myboard_iMX6Q_1GB/RELEASE_GCC5/ARM/ArmPlatformPkg/PlatformPei/PlatformPeiLib/OUTPUT/./PlatformPeiLib.obj -I/home/test/edk2/ArmPlatformPkg/PlatformPei -I/home/test/Build/myboard_iMX6Q_1GB/RELEASE_GCC5/ARM/ArmPlatformPkg/PlatformPei/PlatformPeiLib/DEBUG -I/home/test/edk2/MdePkg -I/home/test/edk2/MdePkg/Include -I/home/test/edk2/MdePkg/Include/Arm -I/home/test/edk2/MdeModulePkg -I/home/test/edk2/MdeModulePkg/Include -I/home/test/edk2/EmbeddedPkg -I/home/test/edk2/EmbeddedPkg/Include -I/home/test/edk2/ArmPkg -I/home/test/edk2/ArmPkg/Include -I/home/test/edk2/ArmPlatformPkg -I/home/test/edk2/ArmPlatformPkg/Include /home/test/edk2/ArmPlatformPkg/PlatformPei/PlatformPeiLib.c
<command-line>:0:1: error: macro names must be identifiers
Building ... /home/test/edk2/MdeModulePkg/Library/LzmaCustomDecompressLib/LzmaCustomDecompressLib.inf [ARM]
make[1]: Leaving directory '/home/test/Build/myboard_iMX6Q_1GB/RELEASE_GCC5/ARM/ArmPlatformPkg/MemoryInitPei/MemoryInitPeiLib'
make[1]: *** [GNUmakefile:307: /home/test/Build/myboard_iMX6Q_1GB/RELEASE_GCC5/ARM/ArmPlatformPkg/MemoryInitPei/MemoryInitPeiLib/OUTPUT/MemoryInitPeiLib.obj] Error 1

build.py...
 : error 7000: Failed to execute command
	make tbuild [/home/test/Build/myboard_iMX6Q_1GB/RELEASE_GCC5/ARM/ArmPlatformPkg/MemoryInitPei/MemoryInitPeiLib]

build.py...
 : error 7000: Failed to execute command
	make tbuild [/home/test/Build/myboard_iMX6Q_1GB/RELEASE_GCC5/ARM/ArmPlatformPkg/PlatformPei/PlatformPeiLib]
[... 96 lines cut ...]
I/TC: Initialized

U-Boot 2018.05-rc1 (Feb 25 2019 - 02:58:55 -0800)

CPU:   Freescale i.MX6Q rev1.5 996 MHz (running at 792 MHz)
CPU:   Extended Commercial temperature grade (-20C to 105C) at 36C
Reset cause: WDOG
I2C:   ready
DRAM:  1 GiB
PMIC:  PFUZE100 ID=0x10
MMC:   FSL_SDHC: 0, FSL_SDHC: 1
PCI:   pcie phy link never came up
No panel detected: default to HDMI
Display: HDMI (1024x768)
In:    serial
Out:   serial
Err:   serial
Net:   using phy at 6
FEC [PRIME]
Initializing 4KB of memory at 0x10817000 as the global page.
Stashed ethaddr = 7a:56:34:12:63:d0
```

Can anyone help me identify what I am doing wrong in both cases?
Thank you.

dvescovi1, 2019-02-27:
I think the biggest clue is the "Reset cause: WDOG"
There is a i.MX6 register that has bits set that describe why the last reset happened. This is what is displayed by u-boot here.
Could be POR (Power on Reset), software generated or the watchdog timed out and reset the processor.
Looks to me like the watchdog fired and reset the processor. 
If I remember correctly the watchdog timer is unconditionally started on power up, it is up to the early startup code to specifically disarm it.

awietholter, 2019-02-27:
I believe it is not, because the message appears after load the uefi:
```
ERROR: new format image overwritten - must RESET the board to recover
resetting ... 
```
and then resets the board.

dvescovi1, 2019-02-27:
Most cases (but not all) if software needs to trigger a reset it forces  "warm boot" which would show up as a different reset cause... not WDOG.

dvescovi1, 2019-02-27:
Another thing you could test is to see if it happens at (or very near) the 16 second mark after power up.
The WDOG timer defaults to a 16 second time out unless specifically disabled. See chap 70 in the iMX6DQRM.pdf processor reference manual.

awietholter, 2019-02-27:
I searched the message:
```
ERROR: new format image overwritten - must RESET the board to recover
resetting ...
```
And I found that this message forces a reset, calling the function:
do_reset() in the file reset.c.
So I believe that this is a forced reset using WDOG to do it.
But I will keep searching.
Any other ideas?
And, thanks for assist me.

jordanrh1, 2019-02-27:
@awietholter You're on the right track searching for `ERROR: new format image overwritten - must RESET the board to recover`. I think this error is coming from u-boot. The reset is intentional by whatever code is spewing this error. The WDOG is used to reset the system; the reset reason will not give any clues.

awietholter, 2019-02-27:
Thanks @jordanrh1, I believe so too.
Do you have any clue what could be my problem?
I don't know if I generate the uefi as below could generate problems with my board, or should I generate EDK2 with my board, changing the word "Sabre" for "myboard". But in this case, I generate the errors indicating in case 1 above.

```
EDK2_DSC=Sabre_iMX6Q_1GB
EDK2_PLATFORM=NXP/Sabre_iMX6Q_1GB
```
That's why I place both problems (cases) in the same issue.

jordanrh1, 2019-02-27:
This error comes from bootm.c. It looks like there is some kind of memory overlap. Double check the memory map and make sure uefi.fit isn't overlapping with some other region of memory.

```c
no_overlap = (os.comp == IH_COMP_NONE && load == image_start);

        if (!no_overlap && load < blob_end && load_end > blob_start) {
                debug("images.os.start = 0x%lX, images.os.end = 0x%lx\n",
                      blob_start, blob_end);
                debug("images.os.load = 0x%lx, load_end = 0x%lx\n", load,
                      load_end);

                /* Check what type of image this is. */
                if (images->legacy_hdr_valid) {
                        if (image_get_type(&images->legacy_hdr_os_copy)
                                        == IH_TYPE_MULTI)
                                puts("WARNING: legacy format multi component image overwritten\n");
                        return BOOTM_ERR_OVERLAP;
                } else {
                        puts("ERROR: new format image overwritten - must RESET the board to recover\n");
                        bootstage_error(BOOTSTAGE_ID_OVERWRITTEN);
                        return BOOTM_ERR_RESET;
                }
        }
```

awietholter, 2019-02-28:
Hi @jordanrh1 , you are right about memory overlap.
I were calling the wrong address.
After I change the address as below, the uefi starts to load.
```
=> fatload mmc 0:2 0x20A20000/uefi.fit
1902057 bytes read in 102 ms (17.8 MiB/s)
=> bootm 0x20A20000
## Loading kernel from FIT Image at 20a20000 ...
   Using 'conf-1' configuration
   Trying 'uefi' kernel subimage
     Description:  UEFI
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x20a200a4
     Data Size:    1900544 Bytes = 1.8 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x10820000
     Entry Point:  0x10820000
     Hash algo:    sha256
     Hash value:   edd78dc099dd63d30f7b7677665303919eee511cce6b3f1eae22d08c4b59b0ad
   Verifying Hash Integrity ... sha256+ OK
   Loading Kernel Image ... OK
   kernel loaded at 0x10820000, end = 0x109f0000

Starting kernel ...

Debug Serial Port Init
UEFI firmware (version  built at 09:58:06 on Feb 22 2019)
```
But I have to solve now several errors when load the uefi.

About the first case, do you have any ideas why I can't build using my edk2 platform? And not using Sabre platform?
Thank you.

christopherco, 2019-03-08:
Hi @awietholter 

Regarding the first issue, looks like gcc is complaining about a compiler parameter but from your output, the compiler commands look correct. It is very odd.

Could you upload your changes imx-iotcore and imx-edk2-platforms changes to your fork and send me pointers to them? I'd like to see if I can replicate the issue and debug from there.

Also, which environment are you building the uefi in? WSL or native Linux?

christopherco, 2019-04-26:
Hi @awietholter, checking in to see if you are still experiencing the build issues.

awietholter, 2019-04-26:
Hi @christopherco, sorry for late reply.
I still have the same issue. I was relocated to another project and this project was paused. 
Also I have a new PC and have to made the configurations again to restart the project.
I will start today the configuration for the environment into my new PC.
About the environment, I'm using Linux Ubuntu 18.10 (in a virtual machine) to build uefi.
As soon I have able to have same issue, I upload the files.
Thanks.

christopherco, 2019-05-22:
Native Linux Ubuntu 18.10 should work well for building the firmwares.

Going to close this issue for now. If you do still experience the build issue on your new setup, please hit the reopen button and add the logs.

closed 2019-05-22 by christopherco