ms-iot/imx-iotcore #44: Windows Kernel crashing

NovTechEng, 2019-03-08:
Hi, I am trying to get Win10 Iot Core up and running on a custom imx6s board.  So far I have gotten my board to boot to the windows nt kernel, but then it crashes.  
In the DSDT.asl file I have disabled everything but include("Dsdt-iMX6Solo.asl"), include("Dsdt-Gpio.asl"), and include("Dsdt-Sdhc.asl").
I took some time and stepped through the kernel using windbg and it appears that nt_80c15000 exits and then everything crashes and the board reboots. 
I will attach both my windbg output and my serial port output. If anyone has any ideas on how to progress from here that would be great.
Thanks.
[windbgoutput.txt](https://github.com/ms-iot/imx-iotcore/files/2946730/windbgoutput.txt)
[VerboseLog2.txt](https://github.com/ms-iot/imx-iotcore/files/2946732/VerboseLog2.txt)

jordanrh1, 2019-03-08:
Try disabling the PEP (`PEP0`) and the SDHC device. It should bugcheck with `INACCESSIBLE_BOOT_DEVICE`. If it doesn't reach this bugcheck, then something might be going wrong with the timer HAL extension or kernel debugger.

NovTechEng, 2019-03-08:
Alright so I disabled the PEP and SDHC device from the .asl files. I then tried to boot the board, and it again crashed but after running the bugcheck analysis it did show INACCESSIBLE_BOOT_DEVICE.
[bugcheckanalysis.txt](https://github.com/ms-iot/imx-iotcore/files/2947235/bugcheckanalysis.txt)

NovTechEng, 2019-03-08:
I re-enabled PEP and the SDHC device, and am getting the same error from bugcheck.  additionally im getting an error message that says "IOINIT: Built-in driver \Driver\sacdrv failed to initialize with status - 0xC0000037"

jordanrh1, 2019-03-08:
The IOINIT error can be ignored.

If you're getting an `INACESSIBLE_BOOT_DEVICE` bugcheck it means the system can't access the boot volume. When the debugger breaks in at the bugcheck, run `!devnode 0 1` and see what the status of the SDHC device is.

Always include !analyze output when you post about a bugcheck. It is extremely helpful.

NovTechEng, 2019-03-11:
Alright problem is fixed. My main issue was my interrupt number in the SDHC device node was wrong.
Thanks for the help.

closed 2019-03-11 by NovTechEng