ms-iot/imx-iotcore #47: Sabre_iMX6QP_1GB  win10 iot boot failed

cjt0701, 2019-03-13:
hi all
the log as below:

Initializing 4KB of memory at 0x10817000 as the global page.
Stashed ethaddr = 00:00:00:00:00:00
1902057 bytes read in 100 ms (18.1 MiB/s)
## Loading kernel from FIT Image at 20a20000 ...
   Using 'conf-1' configuration
   Verifying Hash Integrity ... OK
   Trying 'uefi' kernel subimage
     Description:  UEFI
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x20a200a4
     Data Size:    1900544 Bytes = 1.8 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x10820000
     Entry Point:  0x10820000
     Hash algo:    sha256
     Hash value:   c3c90a9b5f0ff129535e33f0d6d97e64d6aeafee681a2d5e506f74e7c50f5e65
   Verifying Hash Integrity ... sha256+ OK
   Loading Kernel Image ... OK

Starting kernel ...

Debug Serial Port Init
UEFI firmware (version  built at 23:01:01 on Mar 11 2019)
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Core/Dxe/DxeMain/DEBUG/DxeCore.dll /reloc .text at 0x4F905000 /nocode /noclear
Loading DxeCore at 0x004F904000 EntryPoint=0x004F909E81
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Core/Dxe/DxeMain/DEBUG/DxeCore.dll /reloc .text at 0x4F905000 /nocode /noclear
HOBLIST address in DXE = 0x4EDE4010
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/PCD/Dxe/Pcd/DEBUG/PcdDxe.dll /reloc .text at 0x4EF70000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/Acpi/AcpiTableDxe/AcpiTableDxe/DEBUG/AcpiTableDxe.dll /reloc .text at 0x4EF69000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/SmbiosDxe/SmbiosDxe/DEBUG/SmbiosDxe.dll /reloc .text at 0x4EF62000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Core/RuntimeDxe/RuntimeDxe/DEBUG/RuntimeDxe.dll /reloc .text at 0x4EF9A000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/SecurityStubDxe/SecurityStubDxe/DEBUG/SecurityStubDxe.dll /reloc .text at 0x4EF1A000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/ArmPkg/Drivers/ArmGic/ArmGicDxe/DEBUG/ArmGicDxe.dll /reloc .text at 0x4EF14000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/SerialDxe/SerialDxe/DEBUG/SerialDxe.dll /reloc .text at 0x4EF0F000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/EmbeddedPkg/SimpleTextInOutSerial/SimpleTextInOutSerial/DEBUG/SimpleTextInOutSerial.dll /reloc .text at 0x4EF09000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/EmbeddedPkg/ResetRuntimeDxe/ResetRuntimeDxe/DEBUG/Reset.dll /reloc .text at 0x4EF95000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/EmbeddedPkg/RealTimeClockRuntimeDxe/RealTimeClockRuntimeDxe/DEBUG/RealTimeClock.dll /reloc .text at 0x4EF8F000 /nocode /noclear
InitializeRealTimeClock: using default timezone/daylight settings
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/EmbeddedPkg/MetronomeDxe/MetronomeDxe/DEBUG/MetronomeDxe.dll /reloc .text at 0x4EF04000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/Platform/Microsoft/Drivers/SdMmcDxe/SdMmcDxe/DEBUG/SdMmcDxe.dll /reloc .text at 0x4EEF5000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/Silicon/NXP/iMXPlatformPkg/Drivers/SdhcDxe/SdhcDxe/DEBUG/SdhcDxe.dll /reloc .text at 0x4EEEE000 /nocode /noclear
SDHC4:Error detected
SDHC4:WaitForCmdResponse() failed. Device Error
SdMmc[E]:HostExt->SendCommand( CMD8, 0x000001AA) failed. Device Error (SdhcSendCommandHelper: /root/imx-edk2-platforms/Platform/Microsoft/Drivers/SdMmcDxe/Protocol.c, 408)
SdMmc[E]:Send no-data command failed. Device Error (SdhcSendCommand: /root/imx-edk2-platforms/Platform/Microsoft/Drivers/SdMmcDxe/Protocol.c, 453)
SDHC4:Error detected
SDHC4:WaitForCmdResponse() failed. Device Error
SdMmc[E]:HostExt->SendCommand( CMD5, 0x00000000) failed. Device Error (SdhcSendCommandHelper: /root/imx-edk2-platforms/Platform/Microsoft/Drivers/SdMmcDxe/Protocol.c, 408)
SdMmc[E]:Send no-data command failed. Device Error (SdhcSendCommand: /root/imx-edk2-platforms/Platform/Microsoft/Drivers/SdMmcDxe/Protocol.c, 453)
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/Silicon/NXP/iMX6Pkg/Drivers/PciEmulation/PciEmulation/DEBUG/PciEmulation.dll /reloc .text at 0x4EEE8000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/DevicePathDxe/DevicePathDxe/DEBUG/DevicePathDxe.dll /reloc .text at 0x4EEDC000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/HiiDatabaseDxe/HiiDatabaseDxe/DEBUG/HiiDatabase.dll /reloc .text at 0x4EEC5000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/Variable/RuntimeDxe/VariableRuntimeDxe/DEBUG/VariableRuntimeDxe.dll /reloc .text at 0x4EF84000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/Acpi/AcpiPlatformDxe/AcpiPlatformDxe/DEBUG/AcpiPlatform.dll /reloc .text at 0x4EEC0000 /nocode /noclear
remove-symbol-file /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/Acpi/AcpiPlatformDxe/AcpiPlatformDxe/DEBUG/AcpiPlatform.dll 0x4EEC0000
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/Silicon/NXP/iMX6Pkg/Drivers/PlatformSmbiosDxe/PlatformSmbiosDxe/DEBUG/PlatformSmbiosDxe.dll /reloc .text at 0x4EEBE000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/ArmPkg/Drivers/CpuDxe/CpuDxe/DEBUG/ArmCpuDxe.dll /reloc .text at 0x4EEB6000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/CapsuleRuntimeDxe/CapsuleRuntimeDxe/DEBUG/CapsuleRuntimeDxe.dll /reloc .text at 0x4EF7E000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/MonotonicCounterRuntimeDxe/MonotonicCounterRuntimeDxe/DEBUG/MonotonicCounterRuntimeDxe.dll /reloc .text at 0x4EF79000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/Silicon/NXP/iMX6Pkg/Drivers/TimerDxe/TimerDxe/DEBUG/iMX6TimerDxe.dll /reloc .text at 0x4EEB0000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/SetupBrowserDxe/SetupBrowserDxe/DEBUG/SetupBrowser.dll /reloc .text at 0x4EE9C000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/BdsDxe/BdsDxe/DEBUG/BdsDxe.dll /reloc .text at 0x4EE89000 /nocode /noclear
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/SecurityPkg/Tcg/Tcg2Dxe/Tcg2Dxe/DEBUG/Tcg2Dxe.dll /reloc .text at 0x4EE77000 /nocode /noclear
Tpm2AcpiControlAreaInit: About to write control area.
WARNING: There is mismatch of supported HashMask (0x3 - 0x0) between modules
[... 202 lines cut ...]
that are linking different HashInstanceLib instances!
WARNING: There is mismatch of supported HashMask (0x3 - 0x0) between modules
that are linking different HashInstanceLib instances!
WARNING: There is mismatch of supported HashMask (0x3 - 0x0) between modules
that are linking different HashInstanceLib instances!
WARNING: There is mismatch of supported HashMask (0x3 - 0x0) between modules
that are linking different HashInstanceLib instances!
Data.load.elf /root/Build/Sabre_iMX6QP_1GB/RELEASE_GCC5/ARM/ShellPkg/Application/Shell/Shell/DEBUG/Shell.dll /reloc .text at 0x4B86E000 /nocode /noclear
UEFI Interactive Shell v2.2
EDK II
UEFI v2.70 (NXP, 0x00000001)
Mapping table
      FS0: Alias(s):HD0b:;BLK1:
          VenHw(AAFB8DAA-7340-43AC-8D49-0CCE14812489,04000000)/eMMC(0x0)/HD(1,MB
R,0xF2203AB7,0x5000,0xFA000)
     BLK0: Alias(s):
          VenHw(AAFB8DAA-7340-43AC-8D49-0CCE14812489,04000000)/eMMC(0x0)
     BLK2: Alias(s):
          VenHw(AAFB8DAA-7340-43AC-8D49-0CCE14812489,04000000)/eMMC(0x0)/HD(2,MB
R,0xF2203AB7,0x12C000,0xD64000)
Press ESC in 1 seconds to skip startup.nsh or any other key to continue.
Shell> 
Shell> 
Shell> 
Shell>

hbeberman, 2019-03-14:
Hi @cjt0701 
It looks like UEFI is failing to see your SD card device, but it was available to SPL and U-Boot earlier. I'm unable to reproduce your issue with my MCIMX6QP-SDB (SCH-28857 REV B) with both the firmware checked into the [Sabre_iMX6QP_1GB package](https://github.com/ms-iot/imx-iotcore/tree/public_preview/build/board/Sabre_iMX6QP_1GB/) folder, and built from a clean checkout of the firmware sources.

- What board are you using exactly?
- If it's the same board as mine, are you using the SD2 or SD3 port for your SD card?
- Have you tried the firmware in https://github.com/ms-iot/imx-iotcore/tree/public_preview/build/board/Sabre_iMX6QP_1GB
- Does any output appear when you unplug and replug the SD card while at the shell?
- Can you try a different SD card?

cjt0701, 2019-03-16:
> Hi @cjt0701
> It looks like UEFI is failing to see your SD card device, but it was available to SPL and U-Boot earlier. I'm unable to reproduce your issue with my MCIMX6QP-SDB (SCH-28857 REV B) with [...]

cjt0701, 2019-03-16:
hi @hbeberman 
 the default sd card is sd3，but i reconfig the dsc uefi，my board sd card is sd2，it still have issue. log as below
SDHC2:Error detected
SDHC2:WaitForCmdResponse() failed. Device Error
mmc0(sd2),mmc1(sd3),mmc2(emmc)

is there  a issue about hardware ?
my board is the  Sabre_iMX6QP_1GB,

CPU:   Freescale i.MX6QP rev1.1 996 MHz (running at 792 MHz)
CPU:   Automotive temperature grade (-40C to 125C) at 22C
Reset cause: POR
Board: MX6-SabreSD
I2C:   ready
DRAM:  1 GiB
PMIC:  PFUZE100 ID=0x10
MMC:   FSL_SDHC: 0, FSL_SDHC: 1, FSL_SDHC: 2
auto-detected panel HDMI
Display: HDMI (1024x768)
In:    serial
Out:   serial
Err:   serial
Net:   FEC [PRIME]
Error: FEC address not set.

the piublic_preview firmware i can not try,the deflaut SDHC is deferent ,my board is sd2.

cjt0701, 2019-03-18:
hi @hbeberman
the system has booted from the sd2,the hdmi is working ,but the usb and  Ethernet interface is not working.

hbeberman, 2019-03-18:
Your first log shows that your board isn't fused with a MAC address, so Ethernet is unable to function.
You can fuse an Ethernet address onto the board using the fuse command on bank 4 works 2 and 3 in U-Boot shell.
```
=> env print ethaddr
ethaddr=00:06:9f:15:07:18
=> fuse read 4 2
Reading bank 4:
Word 0x00000002: 9f150718
=> fuse read 4 3
Reading bank 4:
Word 0x00000003: 00000006
```
You can stop in U-Boot shell by renaming UEFI.fit in the EFI partition of your SD card.

The OTG USB works best when a hub is plugged in during boot. After booting with a hub I'm able to use a keyboard once IoT Core is booted.

cjt0701, 2019-03-19:
hi @hbeberman
thank you .
the Ethernet and otg usb have worked. Is there any help about host usb?

cjt0701, 2019-03-19:
hi @hbeberman 
thank you 
the Ethernet and otg usb have worked.

closed 2019-03-19 by hbeberman