ms-iot/imx-iotcore #55: Problem while building an Image

Perosperos, 2019-04-05:
Hello,
I am a very beginner with IoT Core, so maybe anyone can help me out.

I usw WIndows 10 Pro and installed WIndows ADK, WDK, WDK, Visual Studio 2017 Enterprise complety new.

I followed the instruction on: https://github.com/ms-iot/imx-iotcore.

At FFU Generation
1. Launch Visual Studio 2017 as Administrator.
2. Open the solution iMXPlatform.sln (imx-iotcore\build\solution\iMXPlatform).
3. Change the build type from Debug to Release. Change the build flavor from ARM to ARM64 if building for iMX8.
4. To build press Ctrl-Shift-B or choose Build -> Build Solution from menu. This will compile all driver packages then generate the FFU.

At Step 4 while compiling i got some errors:

[image]

Can maybe someone give me some tips?

Thanks!

christopherco, 2019-04-06:
Hi @Perosperos 

From the error, looks like TrEEGenService.h header is missing. This header is part of a git submodule called OpteeCalls. It is located at imx-iotcore/driver/TrEE/OpteeCalls/inc/. Please check that this folder is populated. 

If not, do a
```
git submodule init
git submodule update
```
and this should fetch the OpteeCalls submodule.

christopherco, 2019-04-25:
@Perosperos checking in on this. Were you able to get the FFU to build?

Perosperos, 2019-04-25:
Hello !
thanks for the help,

all works fine!

Thanks a lot :)

closed 2019-04-25 by Perosperos