ms-iot/imx-iotcore #81: win10iot for imx8mq

cjt0701, 2019-05-10:
hi all
I have genetated the ffu,but i how to make a sd card boot image for imx8mq.The dash board has only support imx6/7.

christopherco, 2019-05-10:
The dashboard has not been updated to reflect i.mx8 but it is the same option as the imx6/7.

cjt0701, 2019-05-10:
hi @christopherco
Uboot can't start, do not have any log.I am sure the boot config is sd card.The sd clk  pin has output。

christopherco, 2019-05-10:
Which iMX8M board are you using? Also what is the baud rate of your serial debug console?

dvescovi1, 2019-05-11:
just rebuilt i.MX8M EVK image … booted fine.
Rebuilt firmware and rebuilt image … also booted fine.

cjt0701, 2019-05-11:
Hi christopherco and dvescovi1 

the board has booted,but the hdmi output resolution is only 720P.How do i set the output resolution with 1080P?

neilsh-msft, 2019-05-13:
You'd need to modify U-Boot to set the correct timing for 1080P.  You'll need to follow up on the NXP iMX8 community support https://community.nxp.com/community/imx/content.

christopherco, 2019-05-22:
Closing issue for now. If you have more questions or issues, feel free to reopen if it along the same topic or post a new issue.

closed 2019-05-22 by christopherco