ms-iot/imx-iotcore #84: IoT Core as an edge device?

dvescovi1, 2019-05-23:
I am wondering if there is capability or a road map to using imx-iot core as an edge device?
I see the IoT edge run time is supported on AMD64 (desktop Windows) and some armv7 devices … I guess mostly running Linux. Wondering if ARM64 under Windows IoT core is in the (hopefully near) future? Seems like support for containers (Docker?) is required which I am not sure is even supported under the present Windows ARM.
Any insight? Can you point me to any resources?

christopherco, 2019-05-28:
Can you help us understand the business opportunity?  It is on our radar, but there are a lot of dependencies required to light it up.  Details on the business opportunity would help.

dvescovi1, 2019-05-31:
I do not have any immediate business opportunity. It just seems like the logical progression as edge is supported on Linux AMD64, Linux ARM32v7 and Windows AMD64. If anything, it would seem to be more relevant as an IoT core device would more likely be deeply embedded and occasionally disconnected (i.e. a drone). Looks like [AMD64 IoT core may be supported](https://docs.microsoft.com/en-us/azure/iot-edge/support#operating-systems) but Windows IoT core ARM64 is missing.

closed 2019-05-31 by dvescovi1