ms-iot/imx-iotcore #92: Build ffu

Cmac2016, 2019-06-01:
When I try to build this I get the following error:

Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB3073	The command "
      set Configuration=Release
      set Platform=ARM
      set PkgOutput=C:\Users\chrma\Desktop\NXP\imx-iotcore-public_preview\build\solution\iMXPlatform\Build\ARM\Release\
      set OutDir=C:\Users\chrma\Desktop\NXP\imx-iotcore-public_preview\build\solution\iMXPlatform\Build\ARM\Release\GenerateBSP\
      set OEMName=NXP
      call ..\..\..\tools\BuildBSP.bat iMX6 HummingBoardEdge_iMX6DL_1GB
      call ..\..\..\tools\BuildBSP.bat iMX6 HummingBoardEdge_iMX6Q_2GB
      call ..\..\..\tools\BuildBSP.bat iMX6 HummingBoardEdge_iMX6S_512MB
      REM call ..\..\..\tools\BuildBSP.bat iMX6 RSB4411_iMX6Q_1GB
      call ..\..\..\tools\BuildBSP.bat iMX6 Sabre_iMX6Q_1GB
      call ..\..\..\tools\BuildBSP.bat iMX6 Sabre_iMX6QP_1GB
      call ..\..\..\tools\BuildBSP.bat iMX6 Sabre_iMX6SX_1GB
      REM call ..\..\..\tools\BuildBSP.bat iMX6 SabreLite_iMX6Q_1GB
      call ..\..\..\tools\BuildBSP.bat iMX6 UdooNeo_iMX6SX_1GB
      call ..\..\..\tools\BuildBSP.bat iMX6 VAB820_iMX6Q_1GB
      call ..\..\..\tools\BuildBSP.bat iMX7 ClSomImx7_iMX7D_1GB
      call ..\..\..\tools\BuildBSP.bat iMX7 Sabre_iMX7D_1GB
    " exited with code 1.	GenerateBSP	C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets	49	
Error		ThreadId3788 Imaging!ValidateInput: Failed to load Feature Manifest XML file C:\Users\chrma\Desktop\NXP\imx-iotcore-public_preview\build\\board\HummingBoardEdge_iMX6Q_2GB\MergedFMs\HummingBoardEdge_iMX6Q_2GB_DeviceFM.xml:FeatureAPI!ValidateAndLoad: FeatureManifest file was not found: C:\Users\chrma\Desktop\NXP\imx-iotcore-public_preview\build\\board\HummingBoardEdge_iMX6Q_2GB\MergedFMs\HummingBoardEdge_iMX6Q_2GB_DeviceFM.xml	GenerateTestFFU	C:\Users\chrma\Desktop\NXP\imx-iotcore-public_preview\build\solution\iMXPlatform\GenerateTestFFU\imageapp	1	
Error		ThreadId3788 	FeatureAPI!ValidateAndLoad: FeatureManifest file was not found: C:\Users\chrma\Desktop\NXP\imx-iotcore-public_preview\build\\board\HummingBoardEdge_iMX6Q_2GB\MergedFMs\HummingBoardEdge_iMX6Q_2GB_DeviceFM.xml	GenerateTestFFU	C:\Users\chrma\Desktop\NXP\imx-iotcore-public_preview\build\solution\iMXPlatform\GenerateTestFFU\imageapp	1	
Error	MSB3073	The command "
      set Configuration=Release
      set Platform=ARM
      set PkgOutput=C:\Users\chrma\Desktop\NXP\imx-iotcore-public_preview\build\solution\iMXPlatform\Build\ARM\Release\
      set OutDir=C:\Users\chrma\Desktop\NXP\imx-iotcore-public_preview\build\solution\iMXPlatform\Build\ARM\Release\GenerateTestFFU\
      call GenerateFFU.bat
    " exited with code 1.	GenerateTestFFU	C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets	49	

I did verify I was able to build previous versions.

christopherco, 2019-06-06:
I just did a completely fresh setup starting from a new "Windows 10 dev environment" Quick Create HyperV VM and was not able to reproduce these errors.

From looking through the log, it looks like you are using the very latest VS 2019 and SDK/WDK/ADK (build 18362 a.k.a. version 1903). Our instructions expect to use VS2017 and SDK/WDK/ADK version 1809 (build 17763) since these versions match the Windows IoT Core 1809 LTSC version. It is possible that the latest VS or Kits could have caused some new build errors. I'll set up a new environment and try these out.

In the meantime if you would like to try and replicate the HyperV setup mentioned above, here are the exact steps I did:
1. Open Hyper-V Manager
2. Select "Quick Create" -> Windows 10 Dev environment
3. Run through all the Windows installer setup. Eventually you will be taken to a desktop. This image comes pre-installed with VS2017 Community and the 17763 SDK (a.k.a. version 1809)
2. Run Visual Studio 2017 installer. It will prompt you to Update.
3. Under installed products, select Visual Studio Community 2017 -> More -> Modify and check these options
    * VC++ 2017 version 15.9 v14.16 Libs for Spectre (ARM)
    * VC++ 2017 version 15.9 v14.16 Libs for Spectre (ARM64)
    * VC++ 2017 version 15.9 v14.16 Libs for Spectre (X86 and x64)
    * Visual C++ compilers and libraries for ARM
    * Visual C++ compilers and libraries for ARM64 
4. Downloaded and installed the Windows ADK version 1809. Needed to navigate to the [Other ADK downloads page](https://docs.microsoft.com/en-us/windows-hardware/get-started/adk-install) since the most recent one is version 1903.
5. Downloaded and installed the Windows WDK version 1809. Needed to navigate to the [Other WDK downloads page](https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads)
     * As part of the install, made sure to install the WDK Extension for Visual Studio.
6. Downloaded the "Windows 10 IoT Core Packages – Windows 10 IoT Core - version 1809 - January Refresh (LTSC)" from the [Windows IoT Core Downloads Page](https://www.microsoft.com/en-us/software-download/windows10IoTCore#!)
7. Opened the .iso and install Windows_10_IoT_Core_ARM_Packages.msi
8. Installed Git for Windows from the [Git download page](https://git-scm.com/downloads)
9. Opened Administrator command prompt (not Powershell) -> Navigate to new workspace directory -> git clone https://github.com/ms-iot/imx-iotcore.git
10. Navigate to the folder imx-iotcore\build\tools and launch StartBuildEnv.bat
11. In the new window that pops up, run SetupCertificate.bat to install the test certificates. Then close this window
12. In the administrator command prompt, go back to the repo and checkout out the submodules using
     * git submodule init
     * git submodule update
13. Launch Visual Studio 2017 as Administrator
14. File -> Open -> Project/Solution... -> Navigate to imx-iotcore\build\solution\iMXPlatform.sln 
15. Select "Release" from the Solution Configuration.
16. Build -> Build Solution

Cmac2016, 2019-06-06:
Thank you. I will give this a try. Just as a note I can compile older versions of this repo with no problems with my current config.
Initially I did have 2017 but when I uninstalled it(thought I maybe had a corruption) I updated to vs2019.

Cmac2016, 2019-06-06:
Unfortunately I cannot try with your method.  I am running Windows 10 Home and Hyper-V appears to not be available on Home

kiwikissling, 2019-06-14:
I'm getting a similar problem, from cmac2016's log:

> 36>imageapp INFO ThreadId6536 Imaging: Reading the FM XML file 'C:\Users\chrma\imx-iotcore\build\board\HummingBoardEdge_iMX6Q_2GB\MergedFMs\HummingBoardEdge_iMX6Q_2GB_DeviceFM.xml'...
> 36>imageapp : error : ThreadId6536 Imaging!ValidateInput: Failed to load Feature Manifest XML file C:\Users\chrma\imx-iotcore\build\board\HummingBoardEdge_iMX6Q_2GB\MergedFMs\HummingBoardEdge_iMX6Q_2GB_DeviceFM.xml:FeatureAPI!ValidateAndLoad: FeatureManifest file was not found: C:\Users\chrma\imx-iotcore\build\board\HummingBoardEdge_iMX6Q_2GB\MergedFMs\HummingBoardEdge_iMX6Q_2GB_DeviceFM.xml
> 36>imageapp INFO ThreadId6536 Imaging: See HummingBoardEdge_iMX6Q_2GB_TestOEMInput.xml.Release.cbs.log for details.
> 36>imageapp : error : ThreadId6536 FeatureAPI!ValidateAndLoad: FeatureManifest file was not found: C:\Users\chrma\imx-iotcore\build\board\HummingBoardEdge_iMX6Q_2GB\MergedFMs\HummingBoardEdge_iMX6Q_2GB_DeviceFM.xml

Ref ^^^, I am getting the same error in my build, for the same board. The FeatureMerge appears to run, but the `...\imx-iotcore\build\board\HummingBoardEdge_iMX6Q_2GB\MergedFms` folder is empty.

I'm using VS2017 and have removed and reinstalled the ADK, WDK, and SDK several times. The problem must be environmental as I've previously built this project (and earlier versions) successfully in the past. It's possible the build problem first happened when I updated the ADK to 1903. I've since gone through the guidance on this repo's `Readme.md` very carefully, to without success.

christopherco, 2019-06-14:
I have replicated the issue and am in the process of root causing. It occurs if you have ever installed the 1903 ADK. Uninstalling and reinstalling 1809 ADK does not fix the issue.

There was a change to the ADK tools between 1809 -> 1903 that causes the FeatureMerger.exe to throw an exception when validating the input cabs. These tools to build the cabs are found in `C:\Program Files (x86)\Windows Kits\10\tools\bin\i386` and when you uninstall the ADK, apparently the 1903 versions of the tools still stay around.

As an experiment, I manually copied in an 1809 version of this folder. Then I cleaned and rebuilt my solution so the cabs get regenerated. I was able to build the solution again just like before.

I have not found a reliable way to roll back to the 1809 versions. And I am still in the process of isolating which tool in particular is the root cause of the exception.

kiwikissling, 2019-06-14:
@christopherco , that's great news. You're about 2 weeks ahead of me and this knowledge has saved me lots of pain!

I notice that [Customization Guidance](https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/set-up-your-pc-to-customize-iot-core) directs us to the [Download ADK](Download and install the Windows ADK) page which leads with version 1903. I my case, attempting to bring tooling up to latest versions was a wrong, wrong, wrong thing to do!

christopherco, 2019-06-17:
Thanks for pointing the URL out. There is a small note that says the versions must match but we really should just point the URL at the 1809 version instead of the latest ADK, which is now 1903. I'll talk with the doc owners to get that updated.

Also, I have put together steps that cleaned out and reset my environment. Please try them out and let me know if it works for you:
1. Remove all the currently built CABs from imx-iotcore folder. Easiest way to do this is to perform a `git clean -xdf` inside the imx-iotcore workspace. This will delete all files not tracked by git.
   - Alternatively, you can manually delete the `imx-iotcore\build\solution\iMXPlatform\Build` folder.
2. Uninstall your current ADK.
3. Uninstall your current WDK. After both the ADK and WDK are uninstalled, you should have an empty `C:\Program Files (x86)\Windows Kits\10\Tools\bin\` folder.
4. If you have VS2019 installed,
   - This project requires using the 1809 Kits which are only supported with VS2017. If you are using VS2019, it is recommended to uninstall the SDK and VS2019. If you need to keep VS2019, I believe you can install VS2017 side-by-side with VS2019 but I have not personally tested this configuration.
   - Download and install VS2017. Be sure to select the proper ARM VC++ options during install.
   - VS2017 should install the 1809 SDK. If it does not, you can manually download it from the msdn website.
4. Download and install the 1809 WDK.
5. Download and install the 1809 ADK
6. Open imx-iotcore solution and rebuild the FFU

kiwikissling, 2019-06-17:
@christopherco, Thanks for this guidance.  Following each step...
Step 1: Done. Manually deleted the folder
Step 2: Done.
Step 3. **Issue**. Uninstalled the WDK ok. The folder `C:\Program Files (x86)\Windows Kits\10\Tools\bin\` **wasn't** empty. There was a `..\bin\i386` folder .. which I deleted so `..\bin` would become empty
Step 4: I'm using VS2017 so didn't need to do anything here
Step 5: Done.
Step 6: Done.
Step 7: Pulled repo, did submodule update, and did StartBuildEnv.bat and SetupCertificate.bat steps just be sure. The build failed with 19 errors like this one:

> 20>wm8962codec.vcxproj -> C:\Projects\Github\Microsoft\imx-iotcore\build\solution\iMXPlatform\Build\ARM\Release\Wm8962codec\Wm8962codec.dll
> 20>Done Adding Additional Store
> 20>Successfully signed: C:\Projects\Github\Microsoft\imx-iotcore\build\solution\iMXPlatform\Build\ARM\Release\Wm8962codec\Wm8962codec.dll
> 20>
> 20>C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.common.targets(1750,5): error MSB3721: The command ""C:\Program Files (x86)\Windows Kits\10\Tools\bin\i386\pkggen.exe" wm8962codec.wm.xml /config:"C:\Program Files (x86)\Windows Kits\10\Tools\bin\i386\pkggen.cfg.xml" /output:"C:\Projects\Github\Microsoft\imx-iotcore\build\solution\iMXPlatform\Build\ARM\Release\Wm8962codec\Wm8962codec CBS\\" /version:1.0.0.0 /build:fre /cpu:ARM /variables:"HIVE_ROOT=C:\Program Files (x86)\Windows Kits\10\\CoreSystem\10.0.17763.0\arm;WIM_ROOT=C:\Program Files (x86)\Windows Kits\10\\CoreSystem\10.0.17763.0\arm;_RELEASEDIR=C:\Projects\Github\Microsoft\imx-iotcore\build\solution\iMXPlatform\Build\ARM\Release\Wm8962codec\;TARGETNAME=Wm8962codec;TARGETEXT=.dll;DRIVER_DEST=drivers\umdf;OEMNAME=NXP" /universalbsp" exited with code 1.
> 20>Done building project "wm8962codec.vcxproj" -- FAILED.

I can see the build is expecting an `\i386` folder (and contents) which are not present.. after I deleted them in Step 3.

Can you suggest a way forward? Which tool populates the `\i386` folder?

christopherco, 2019-06-18:
For 1809, the WDK is responsible for populating the `\i386` folder. As a sidenote, the 1903 WDK no longer populates the `\i386` folder (and likely now no longer deletes it) and instead the ADK populates that folder.

I'm a bit concerned that there are still files in the `Tools\bin` folder after uninstalling the WDK.

Please confirm that your setup installers are the 1809 versions of the WDK and ADK (10.1.17763.1)
[image]

kiwikissling, 2019-06-18:
Yes, I have the correct WDK and ADK versions. I also have a solution (of sorts)!

Last week I started building a VM, and now have this in a working state - vs2017, correct sdk/adk/wdk, iot-imxcore etc and can build the FFU. Just now, I copied the `..bin\i386\` folder and contents back to my physical laptop, and the FFU now appears to be building ok here, ie none of the errors shown above.

I too am a bit concerned about 1903 not working for us. Thanks for raising the version issue with the documentation owners. 

So I'm good for now. Next step is to get our o/s and app booting from the Hummingboards eMMC. Fun times!