ms-iot/imx-iotcore #94: GPIO 6

NovTechEng, 2019-06-05:
On our IMX6S board, we are attempting to use GPIO6 as I2C3_SDA (Alt 5).  It is correctly muxed up until the windows kernel where for some reason the muxing changes (instead of 0x15 there is now 0x11).  We attempted to fix this by manually muxing the gpio in the imxgpio driver but this now leads to the board crashing. We have had inconsistent successful boots with only the minimal drivers (Dsdt-iMX6Solo, Dsdt-Gpio, and Dsdt-Sdhc), but once the ethernet driver is added it always crashes.  I assume we need to find another way to stop windows from messing up the muxing but I am not sure what is causing that in the first place.

I would appreciate any help.  Please let me know of any relevant additional information that I could give.

christopherco, 2019-06-05:
I believe you are seeing a side-effect of a workaround for an NXP errata: ERR006687
https://www.nxp.com/docs/en/errata/IMX6SDLCE.pdf

The workaround listed for the errata is to configure GPIO6 to output ENET interrupts. We implement the workaround in the PEP driver.

https://github.com/ms-iot/imx-iotcore/blob/public_preview/driver/power/imx6pep/sys/mx6dpm.cpp#L1412-L1422

Depending on your requirements, one option would be to not load the PEP driver. This would mean you don't get runtime device and cpu power management (which I'm not sure has been implemented on the iMX6S anyways?) and you don't hit this enet workaround flow.

To remove the PEP you can delete the PEP0 device node in your DSDT. For example: https://github.com/ms-iot/imx-edk2-platforms/blob/6119c0345d6192857cacf9a8bbd811f8ab7c9be8/Silicon/NXP/iMX6Pkg/AcpiTables/Dsdt-Platform.asl#L82

NovTechEng, 2019-06-07:
I deleted the PEP0 device node in my DSDT but the PEP driver is still being loaded.  Additionally, If I wish to use the GPIO6 pin for i2c will I be able to also use ethernet?

NovTechEng, 2019-06-07:
Commenting out the workaround flow in the PEP driver results in the board crashing whenever the Ethernet driver is enabled (similar to before when the muxing was being manually changed in the gpio driver).  Is there anyway to enable ethernet and still use gpio 6 for i2c with the current ethernet driver?  (I could be wrong but the errata seems to imply that ethernet can still be used without the workaround it will just have reduced performance).

christopherco, 2019-06-07:
I too was under the expectation that ethernet can still be used without the workaround. You would just not be able to wake the system from a sleep state, which depending on your requirements could be an acceptable limitation.

Will investigate this and get back to you.

jordanrh1, 2019-06-07:
You can't just comment out the workaround - you have to disable low power states. If you uncomment the workaround without disabling low power states, the system will not respond to ethernet interrupts in a timely manner, leading to issues. You can disable low power states by running

```
powercfg.exe /setacvalueindex SCHEME_CURRENT SUB_PROCESSOR IdleDisable 1
powercfg.exe /setdcvalueindex SCHEME_CURRENT SUB_PROCESSOR IdleDisable 1
powercfg.exe /setactive SCHEME_CURRENT
```
Try removing the GpioInt resource in the ENET device node:

https://github.com/ms-iot/imx-edk2-platforms/blob/f0a7a949cc70a817fa97a3e568774b0cece15f27/Platform/NXP/Sabre_iMX6QP_1GB/AcpiTables/Dsdt-Enet.asl#L38

NovTechEng, 2019-06-07:
Where should I be placing/entering that code?

I removed the GpioInt with both the PEP0 node removed when manually undoing the workaround and the board still crashed with the ethernet driver enabled

NovTechEng, 2019-06-10:
I did some additional messing with the code.  Removing the PEP0 device node causes the PEP driver to still load but it causes a fatal system error.

[PEP error.txt](https://github.com/ms-iot/imx-iotcore/files/3272880/PEP.error.txt)

christopherco, 2019-06-11:
Hi @NovTechEng 

Finished my investigation.

I was mistaken earlier - the PEP is a special driver and is not loaded through the traditional ACPI way. The only way to "disable" the PEP is to not include it in the image at all. Simply deleting the DSDT node only removes its resources but the PEP will still attempt to load.

In order to achieve the behavior you are looking for, you need to do 2 things:
1. Delete the mx6pep.sys from your MainOS partition. You can do this by either: 
   * deleting mx6pep.sys from your SD card located at `MainOS:\Windows\System32\DriverStore\FileRepository\mx6pep.inf_arm_89caa9f38d96a663\mx6pep.sys`
Or
   * deleting the `NXP.iMX.mx6pep.cab` entry from your board's InputFMs file. For example, delete this line and rebuild/reflash the FFU but for your board:
https://github.com/ms-iot/imx-iotcore/blob/beb34de146b2ee44f889fca608f087e942ee2656/build/board/HummingBoardEdge_iMX6S_512MB/InputFMs/HummingBoardEdge_iMX6S_512MB_DeviceFM.xml#L13

2. As Jordan mentioned, you must also delete the GpioInt resource in your ENET DSDT node otherwise you will hang during boot. The Dsdt-Enet.asl file could either be in your Platform/AcpiTables folder or the default Silicon/NXP/iMX6Pkg/AcpiTables folder. I recommend that you create a Dsdt-Enet.asl file in your Platform/Board area so you can control what goes in it.

Both of these together allow me to still boot my Hummingboard iMX6S system and GPIO6 is its power-on default.
```
kd> !dd [uc] 0x20E0234 L1

# 20e0234 00000005
```

NovTechEng, 2019-06-11:
Awesome! That worked. Thank you for the help.

closed 2019-06-11 by NovTechEng