ms-iot/imx-iotcore #127: How to deploy and run applications on NXPEVK_iMX8M_Mini_2GB platform

ShiaujiuanChen, 2019-08-21:
I'm trying to run the applications on iMX8M Mini EVK board, and having some problems

- Deploy HelloWorld UWP application, and show output as follows:

1>------ Deploy started: Project: HelloWorld, Configuration: Debug ARM64 ------
1>Error: Unable to connect to the Microsoft Visual Studio Remote Debugger named '192.168.137.124'.  The Visual Studio 2017 Remote Debugger (MSVSMON.EXE) does not appear to be running on the remote computer. This may be because a firewall is preventing communication to the remote computer. Please see Help for assistance on configuring remote debugging.
========== Build: 0 succeeded, 0 failed, 1 up-to-date, 0 skipped ==========
========== Deploy: 0 succeeded, 1 failed, 0 skipped ==========

- Build ConsoleApplication for ARM64, which prints "Hello World!"

`#include "pch.h"`
`#include <iostream>`

`int main()`
`{`
`    std::cout << "Hello World!\n"; `
`}`

copy the binary to the iMX8M Mini board, and run the command as follows
[image]
no output

Any suggestions what i should try?

christopherco, 2019-08-23:
Just a guess - You might be missing some dlls on the system. The failure signature is no output. Usually the deployment would also copy over any required dlls in your VC runtime. If you manually copy, you must make sure any required dlls are present.

On the target system, can you run depends.exe and target your application? I believe depends.exe should be present in C:\Windows\System32. Depends.exe will dump all the dlls required by your app and indicate any missing ones.

Alternatively you could rebuild your application to statically build in runtime libraries instead of relying on dynamic libs. In VS: 

Project properties -> Under Configuration Properties -> C/C++ -> Code Generation -> Runtime Library -> Select `Multi-threaded (/MT)`

ShiaujiuanChen, 2019-08-26:
Hi, @christopherco ,

Thank you very much for your suggestion!

Now I can deploy HelloWorld application and output ConsoleApplication1.

> * Deploy HelloWorld UWP application, and show output as follows:

Debug -> HelloWorld Properties -> Debug -> Authentication Mode: -> Select **`Universal (Unencrypted Protocol)`**

[image]

> * Build ConsoleApplication for ARM64, which prints "Hello World!"

> On the target system, can you run depends.exe and target your application? I believe depends.exe should be present in C:\Windows\System32. Depends.exe will dump all the dlls required by your app and indicate any missing ones.

```
administrator@MINWINPC C:\Data>ConsoleApplication1.exe

administrator@MINWINPC C:\Data>c:\windows\system32\depends.exe ConsoleApplication1.exe

=== C:\Data\consoleapplication1.exe ===

  Dependencies:
    kernel32.dll
    msvcp140d.dll
    ucrtbased.dll
    vcruntime140d.dll

  ERRORS:
    kernel32.dll (Module not found)
    msvcp140d.dll (Module not found)
    ucrtbased.dll (Module not found)
    vcruntime140d.dll (Module not found)

1 modules loaded, 0 functions resolved, 1 errors.
```

> Project properties -> Under Configuration Properties -> C/C++ -> Code Generation -> Runtime Library -> Select **`Multi-threaded (/MT)`**

```
administrator@MINWINPC C:\Data>ConsoleApplication1.exe
Hello World!

administrator@MINWINPC C:\Data>c:\windows\system32\depends.exe ConsoleApplication1.exe

=== C:\Data\consoleapplication1.exe ===

  Dependencies:
    kernel32.dll

  ERRORS:
    kernel32.dll (Module not found)

1 modules loaded, 0 functions resolved, 1 errors.

administrator@MINWINPC C:\Data>
```

closed 2019-08-26 by ShiaujiuanChen