ms-iot/imx-iotcore #129: Unable to boot SabreSD Quad board

Abdul-0013, 2019-08-23:
The boot is failing with the below prints. 

U-Boot 2018.09-ge52dae0 (Dec 07 2018 - 17:41:28 -0800)

CPU:   Freescale i.MX6Q rev1.2 996 MHz (running at 792 MHz)
CPU:   Automotive temperature grade (-40C to 125C) at 30C
Reset cause: POR
Board: MX6-SabreSD
I2C:   ready
DRAM:  1 GiB
PMIC:  PFUZE100 ID=0x10
MMC:   FSL_SDHC: 0, FSL_SDHC: 1, FSL_SDHC: 2
No panel detected: default to Hannstar-XGA
Display: Hannstar-XGA (1024x768)
In:    serial
Out:   serial
Err:   serial
Net:   FEC [PRIME]
Initializing 4KB of memory at 0x10817000 as the global page.
Stashed ethaddr = fd:e3:02:9f:04:00
1902557 bytes read in 100 ms (18.1 MiB/s)
## Loading kernel from FIT Image at 20a20000 ...
   Using 'conf-1' configuration
   Verifying Hash Integrity ... OK
   Trying 'uefi' kernel subimage
     Description:  UEFI
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x20a200a4
     Data Size:    1900544 Bytes = 1.8 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x10820000
     Entry Point:  0x10820000
     Hash algo:    sha256
     Hash value:   b461e288967bbde9b048004e5ed6637e3f982177d79e934b3a552077c3fc47dd
   Verifying Hash Integrity ... sha256+ OK
   Loading Kernel Image ... OK

Starting kernel ...

Debug Serial Port Init
UEFI firmware (version  built at 16:56:16 on Dec  7 2018)
Data.load.elf /mnt/e/public_preview/Build/Sabre_iMX6Q_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Core/Dxe/DxeMain/DEBUG/DxeCore.dll /reloc .text at 0x4F909000 /nocode /noclear
Loading DxeCore at 0x004F908000 EntryPoint=0x004F90DE79
Data.load.elf /mnt/e/public_preview/Build/Sabre_iMX6Q_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Core/Dxe/DxeMain/DEBUG/DxeCore.dll /reloc .text at 0x4F909000 /nocode /noclear
HOBLIST address in DXE = 0x4EDE4010
Data.load.elf /mnt/e/public_preview/Build/Sabre_iMX6Q_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/PCD/Dxe/Pcd/DEBUG/PcdDxe.dll /reloc .text at 0x4EF70000 /nocode /noclear
Data.load.elf /mnt/e/public_preview/Build/Sabre_iMX6Q_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/Acpi/AcpiTableDxe/AcpiTableDxe/DEBUG/AcpiTableDxe.dll /reloc .text at 0x4EF69000 /nocode /noclear
Data.load.elf /mnt/e/public_preview/Build/Sabre_iMX6Q_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/SmbiosDxe/SmbiosDxe/DEBUG/SmbiosDxe.dll /reloc .text at 0x4EF62000 /nocode /noclear
Data.load.elf /mnt/e/public_preview/Build/Sabre_iMX6Q_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Core/RuntimeDxe/RuntimeDxe/DEBUG/RuntimeDxe.dll /reloc .text at 0x4EF9A000 /nocode /noclear
Data.load.elf /mnt/e/public_preview/Build/Sabre_iMX6Q_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/SecurityStubDxe/SecurityStubDxe/DEBUG/SecurityStubDxe.dll /reloc .text at 0x4EF1A000 /nocode /noclear
Data.load.elf /mnt/e/public_preview/Build/Sabre_iMX6Q_1GB/RELEASE_GCC5/ARM/ArmPkg/Drivers/ArmGic/ArmGicDxe/DEBUG/ArmGicDxe.dll /reloc .text at 0x4EF14000 /nocode /noclear
Data.load.elf /mnt/e/public_preview/Build/Sabre_iMX6Q_1GB/RELEASE_GCC5/ARM/MdeModulePkg/Universal/SerialDxe/SerialDxe/DEBUG/SerialDxe.dll /reloc .text at 0x4EF0F000 /nocode /noclear
Data.load.elf /mnt/e/public_preview/Build/Sabre_iMX6Q_1GB/RELEASE_GCC5/ARM/EmbeddedPkg/SimpleTextInOutSerial/SimpleTextInOutSerial/DEBUG/SimpleTextInOutSerial.dll /reloc .text at 0x4EF09000 /nocode /noclr
Data.load.elf /mnt/e/public_preview/Build/Sabre_iMX6Q_1GB/RELEASE_GCC5/ARM/EmbeddedPkg/ResetRuntimeDxe/ResetRuntimeDxe/DEBUG/Reset.dll /reloc .text at 0x4EF95000 /nocode /noclear
Data.load.elf /mnt/e/public_preview/Build/Sabre_iMX6Q_1GB/RELEASE_GCC5/ARM/EmbeddedPkg/RealTimeClockRuntimeDxe/RealTimeClockRuntimeDxe/DEBUG/RealTimeClock.dll /reloc .text at 0x4EF8F000 /nocode /noclear
InitializeRealTimeClock: using default timezone/daylight settings
Data.load.elf /mnt/e/public_preview/Build/Sabre_iMX6Q_1GB/RELEASE_GCC5/ARM/EmbeddedPkg/MetronomeDxe/MetronomeDxe/DEBUG/MetronomeDxe.dll /reloc .text at 0x4EF03000 /nocode /noclear
Data.load.elf /mnt/e/public_preview/Build/Sabre_iMX6Q_1GB/RELEASE_GCC5/ARM/Platform/Microsoft/Drivers/SdMmcDxe/SdMmcDxe/DEBUG/SdMmcDxe.dll /reloc .text at 0x4EEF4000 /nocode /noclear
Data.load.elf /mnt/e/public_preview/Build/Sabre_iMX6Q_1GB/RELEASE_GCC5/ARM/Silicon/NXP/iMXPlatformPkg/Drivers/SdhcDxe/SdhcDxe/DEBUG/SdhcDxe.dll /reloc .text at 0x4EEED000 /nocode /noclear
SDHC3:Error detected
SDHC3:WaitForCmdResponse() failed. Device Error
SdMmc[E]:HostExt->SendCommand( CMD5, 0x00000000) failed. Device Error (SdhcSendCommandHelper: /mnt/e/public_preview/imx-edk2-platforms/Platform/Microsoft/Drivers/SdMmcDxe/Protocol.c, 408)
SdMmc[E]:Send no-data command failed. Device Error (SdhcSendCommand: /mnt/e/public_preview/imx-edk2-platforms/Platform/Microsoft/Drivers/SdMmcDxe/Protocol.c, 453)
SDHC4:Error detected
SDHC4:WaitForCmdResponse() failed. Device Error
SdMmc[E]:HostExt->SendCommand( CMD8, 0x000001AA) failed. Device Error (SdhcSendCommandHelper: /mnt/e/public_preview/imx-edk2-platforms/Platform/Microsoft/Drivers/SdMmcDxe/Protocol.c, 408)
SdMmc[E]:Send no-data command failed. Device Error (SdhcSendCommand: /mnt/e/public_preview/imx-edk2-platforms/Platform/Microsoft/Drivers/SdMmcDxe/Protocol.c, 453)
SDHC4:Error detected
[... 158 lines cut ...]
WARNING: There is mismatch of supported HashMask (0x3 - 0x0) between modules
that are linking different HashInstanceLib instances!
WARNING: There is mismatch of supported HashMask (0x3 - 0x0) between modules
that are linking different HashInstanceLib instances!
WARNING: There is mismatch of supported HashMask (0x3 - 0x0) between modules
that are linking different HashInstanceLib instances!
WARNING: There is mismatch of supported HashMask (0x3 - 0x0) between modules
that are linking different HashInstanceLib instances!
WARNING: There is mismatch of supported HashMask (0x3 - 0x0) between modules
that are linking different HashInstanceLib instances!
WARNING: There is mismatch of supported HashMask (0x3 - 0x0) between modules
that are linking different HashInstanceLib instances!
WARNING: There is mismatch of supported HashMask (0x3 - 0x0) between modules
that are linking different HashInstanceLib instances!
WARNING: There is mismatch of supported HashMask (0x3 - 0x0) between modules
that are linking different HashInstanceLib instances!
WARNING: There is mismatch of supported HashMask (0x3 - 0x0) between modules
that are linking different HashInstanceLib instances!
WARNING: There is mismatch of supported HashMask (0x3 - 0x0) between modules
that are linking different HashInstanceLib instances!
ConvertPages: failed to find range 10000000 - 10145FFF
Data.load.elf bootmgfw.pdb /reloc .text at 0x4AEEE400 /nocode /noclear
ConvertPages: failed to find range 102000 - 102FFF
#�KDTARGET: Refreshing KD connection
                                     �

christopherco, 2019-08-23:
Could you describe the failure you are seeing? How far into boot do you get?

> #��KDTARGET: Refreshing KD connection

This message is the Windows kernel debugger is attempting to connect over serial, which indicates you are making it into Windows. At this point, debug prints will generally stop so to get further debug info in Windows, you will want to connect Windbg.

Abdul-0013, 2019-08-24:
Hi,

Thank you for your reply.I am not familiar with the windbg. I tried to 
connect the board and i am seeing the below prints:

WARNING: nt overlaps hal_8140e000
WARNING: nt overlaps hal
WARNING: kdcom overlaps nt_81693000
WARNING: kdcom overlaps nt
IOINIT: Built-in driver \Driver\sacdrv failed to initialize with status 
- 0xC0000037
KDTARGET: Refreshing KD connection

I was hoping to see some thing on the LVDS when its booting. But it just 
flashes once in u-boot on power on, then it goes blank.

Regards,
Abdul Ahad
*From:* Christopher Co <notifications@github.com>
*Sent:* Friday, August 23, 2019 10:47PM
*To:* Ms-iot/imx-iotcore <imx-iotcore@noreply.github.com>
*Cc:* Abdul-0013 <abdulahad@iwavesystems.com>, Author 
<author@noreply.github.com>
*Subject:* Re: [ms-iot/imx-iotcore] Unable to boot SabreSD Quad board (#129)

Abdul-0013, 2019-08-26:
Hi,

Some times the below prints are seen in the windbg:

Opened \\.\com4
Waiting to reconnect...
Connected to Windows 10 17763 ARM (NT) Thumb-2 target at (Mon Aug 26 12:56:21.251 2019 (UTC + 5:30)), ptr64 FALSE
Kernel Debugger connection established.
Symbol search path is: srv*
Executable search path is: 
Windows 10 Kernel Version 17763 MP (1 procs) Free ARM (NT) Thumb-2
Built by: 17763.253.armfre.rs5_release_svc_prod1.181231-1749
Machine Name:
Kernel base = 0x8168c000 PsLoadedModuleList = 0x818cf398
System Uptime: 0 days 0:00:00.000
IOINIT: Built-in driver \Driver\sacdrv failed to initialize with status - 0xC0000037
KDTARGET: Refreshing KD connection

Regards,
Abdul Ahad