ms-iot/imx-iotcore #135: Issue #125 fix breaks case handling for non-mount builds

TGoodhew, 2019-09-16:
The changes introduced in #125 break case handling for local cloned builds in WSL via /mnt/c.

The repo is to follow the clones from the firmware documentation and then change to the /mnt/c/<<imx-source>>/build/firmware/HummingBoardEdge_iMX6Q_2GB directory and run make.

dmcilvaney, 2019-09-16:
Thanks, I'll take a look at it.

TGoodhew, 2019-09-16:
Thanks.

TonyG

TGoodhew, 2019-09-16:
Actually I can’t work around by setting the entire directory to case sensitive as that breaks the VS project ☹

TonyG

dmcilvaney, 2019-09-16:
Set the /build/firmware/<board>/u-boot  and /build/firmware/<board>/optee folders to case sensitive, those are the only two which matter.

TGoodhew, 2019-09-16:
Thanks – That appears to have worked.

TonyG

dmcilvaney, 2019-09-16:
Should be fixed in the makefile as well, it snuck past our CI since we run those in native Linux vms.

TGoodhew, 2019-09-16:
Thanks again - Verified the make file fix on my machine

closed 2019-09-16 by TGoodhew