ms-iot/imx-iotcore #146: TA Panic on boot

kiwikissling, 2019-10-02:
I've been working with build `98186ef6` successfully. Today I pulled and built `e7161ba` and have since encountered a TA Panic on boot. 

[image]

Am I missing something in the setup/build process?

neilsh-msft, 2019-10-07:
The RPMB partition on the eMMC needs to be reset.  Apply the following patch, rebuild the OpTEE firmware and boot the board once with the new image.  This should reset the RPMB partition and then the board can boot with the standard firmware.

diff --git a/build/firmware/Common.mk b/build/firmware/Common.mk
index b953401..87abeff 100644
--- a/build/firmware/Common.mk
+++ b/build/firmware/Common.mk
@@ -78,7 +78,7 @@ OPTEE_FLAGS= \
        CFG_SCTLR_ALIGNMENT_CHECK=n \
        CFG_DT=n \
        CFG_NS_ENTRY_ADDR= \
-       CFG_RPMB_RESET_FAT=n \
+       CFG_RPMB_RESET_FAT=y \
        CFG_CORE_HEAP_SIZE=131072 \
        platform-cflags="-fshort-wchar" \
        CFLAGS="-fshort-wchar" \

kiwikissling, 2019-10-22:
Hi @neilsh-msft, thanks for your support. I'm a relative noob in this area, my only experience to-date being building IMX-IOTCore using VS. I have pulled https://github.com/ms-iot/optee_os. Is this the correct project, as I can't find /build/firmware/Common.mk?

kiwikissling, 2019-10-22:
@neilsh-msft, Thanks for your help here. I'm a relative noob at this stuff. My experience to-date has simply been pulling and building the imx-iotcore solution. I found the `CFG_RPMB_RESET_FAT` setting and have changed it as suggested. I now have no idea how to build the firmware, having never done this previously. I have installed GunWin32 Make but I get loads of make errors when making `..\imx-iotcore\build\firmware`

neilsh-msft, 2019-10-22:
Our documentation for building the firmware is published [here](https://github.com/ms-iot/imx-iotcore/blob/public_preview/Documentation/build-arm64-firmware.md).  It has instructions and what tools are necessary for building the firmware.

kiwikissling, 2019-10-23:
Oh wow. What a steep learning curve. This Windows lad has just lost his Linux virginity. Quite a bit of fumbling around. Hope it'll be more fun next time :)

I installed Ubuntu, went through the firmware build guidance and I got as far as installing NXP's code signing tool. I then following the the code signing tool's `Release_Notes.txt` ... but with no luck. Some problem with OpenSSL headers which I'm still looking for. This is turning out to be a big chunk of work just to clear that RPMB store! Will keep trying, but happy to receive any further guidance. Thanks.

neilsh-msft, 2019-10-24:
Unless you're planning on implementing high assurance boot, you shouldn't need to worry about generating signed firmware using NXP's code signing tool.  By default it should just use the default test certificates.  

If you make the makefile change I mentioned above and follow the directions on the building arm firmware page:  https://github.com/ms-iot/imx-iotcore/blob/public_preview/Documentation/build-firmware.md up to the "Deploying firmware to an SD card manually" section you should have a repo in a state that if you build the FFU file in Visual Studio, the resulting FFU should clear the RPMB partition on every boot.  I keep a special SD card with the updated OpTEE that I use to reset RPMB partitions for people in my group who run into this issue.

neilsh-msft, 2019-10-24:
Note: I'm going to be out of office for a week starting tomorrow, so I'll ask someone to keep an eye on this thread.

kiwikissling, 2019-10-30:
Thanks for your help @neilsh-msft. This worked a treat, and I have saved the reset SD card as suggested

closed 2019-10-30 by kiwikissling